ethz-asl/okvis #1: Minor compilation issues on ubuntu 15.10

higerra, 2016-02-06:
First thanks very much for the excellent library! I'm always looking for a SLAM library which provides non-ROS option.

The compilation goes pretty smooth except for two minor issues.
The system environment before compilation: Ubuntu 15.10, gcc 4.9.3, opencv 3.0, Eigen 3.2.7, gflags, glog, Ceres 1.11, Nvidia 352.63, cuda 7.5
Other dependencies are installed by following the guide.

1. In build/brisk/src/brisk_external/src/image-down-sampling.cc, error: 'runtime_error' is not a member of 'std'. Solved this by simply add #include<stdexcept> at the head of this file. This is pretty strange that this source file is located inside the build directory (not from any file in downloaded files).
2. In okvis_cv/include/okvis/implementation/Frame.hpp. Problem occurs in line 120, 153 and 188, all with cv::FeatureDetector and cv::DescriptorExtractor. They are declared as shared_ptr<const cv::FeatureDetector> and shared_ptr<const cv::DescriptorExtractor>, but their member function is called from a non-const function. Also, for the cv::DescritproExtractor, I guess the compute() member function is  not marked as const, so there is an error when trying to call compute() method with shared_ptr<const>.
My current solution is to cast away the const qualifier, for example:

line 188: `  const_cast<cv::DescriptorExtractor*>(extractor_.get())->compute(image_, keypoints_, descriptors_);`

line 120: `  const_cast<cv::FeatureDetector*>(detector_.get())->detect(image_, keypoints_);`

This might not be the desired solution, but it works.

The above are just for compilation, actually I haven't run or use the library.

bhack, 2016-03-18:
Yes i found the same problem.

sleutenegger, 2016-03-23:
Thanks for reporting this. It seems I will have to fix (1) in the BRISK library. (2) is probably linked to a specific version of OpenCV, unfortunately some of the interfaces change sometimes. I will have a look...

bhack, 2016-03-23:
We still cannot use BRISK version in Opencv instead of the external one? /cc @vpisarev