ethz-asl/okvis #6: Failed to get ceres-solver

Richardche, 2016-04-29:
HI, 
    When I make -j8 to build the whole project, it failed in cloning ceres-solver repository. But I can download ceres-solver-1.11.0 via its website. Is there any alternative so that I can build the project without using git clone "https://ceres-solver.googlesource.com/ceres-solver"?

Here is my CMake Error:
CMake Error at /home/richard/okvis/okvis-master/build/ceres/tmp/ceres_external-gitclone.cmake:40 (message):
  Failed to clone repository:
  'https://ceres-solver.googlesource.com/ceres-solver'

SandW, 2016-05-07:
Download the package via the cmd:
`wget http://ceres-solver.org/ceres-solver-1.9.0.tar.gz`
and install it from source.

Set the "USE_SYSTEM_CERES" "ON" in the CMakeLists of OKVIS.

mechaliomar, 2019-04-10:
hello, I have the same problem.Could you explaine more

NILIWEI, 2020-03-29:
sudo gedit /etc/hosts, do address mapping.
change the google source ceres_solver to github source.

super-scyu, 2021-12-24:
modify the cmakelists.txt, line 151, https://github.com/ceres-solver/ceres-solver.git   
and remake

midskymid, 2024-03-10:
Set the "USE_SYSTEM_CERES" "ON" in the CMakeLists of OKVIS and don't forget to change to your ceres version (modify the same   file , line  144 and 148)