ethz-asl/okvis #7: hard to find out the imu noise parameters

learnsor, 2016-04-30:
I managed to compile the okvis code under visual studio 2013. And it worked with the example datasets!  
  But when I tried to run it with my own setup of stereo camera + imu, it just crash. And I thingk it is because of the imu nois parameters.

okvis crash inside "pthread_cond_wait" when adding imu data:
`      if (size >= max_queue_size) {
        pthread_cond_wait(&condition_full_, &mutex_);
      }`

 I dive into kalibr for a few days, I can now calibrate the stereo cameras with IMU.  And I learned a little about Allan avairance . 
It has puzzled me for  a week , but I still cannot figure out the  imu noise parameters for invensense mpu6050. The mpu6050 datasheet does not give me the noise infomation  such as "gyro bias prior",  "gyro drift noise density" 。

Can anyone tell me something about "gyro bias prior",  "gyro drift noise density" of mpu6050?  The mpu6050 datasheet can be found [here](https://www.cdiweb.com/datasheets/invensense/MPU-6050_DataSheet_V3%204.pdf).

learnsor, 2016-05-01:
sorry, I later found that the  crash is because some of the imu timestamp is not  correct。

closed 2016-05-02 by learnsor

David-Willo, 2017-08-22:
Hi, i am also working on a mpu6050 and being very confused about the noise parameters, can you share your config file?