ethz-asl/okvis #17: T_SC

maym86, 2016-07-15:
Is T_SC in config.yaml for a cam0 equivalent to the kalibr output for:

> T_ci:  (imu to cam0): [m]
> [[ 0.0356961  -0.99798199  0.05251415 -0.01277718]
>  [ 0.0821689  -0.04943879 -0.99539142  0.01413488]
>  [ 0.99597895  0.03984663  0.08023831  0.01084148]
>  [ 0.          0.          0.          1.        ]]

or

> T_ic:  (cam0 to imu): [m]
> [[ 0.0356961   0.0821689   0.99597895 -0.01150323]
>  [-0.99798199 -0.04943879  0.03984663 -0.01248458]
>  [ 0.05251415 -0.99539142  0.08023831  0.01387081]
>  [ 0.          0.          0.          1.        ]]

Thanks

Mike

aforster, 2016-07-15:
`S` (for sensor) represents the IMU frame in okvis. So `T_SC = T_IC`

maym86, 2016-07-15:
So 

In the sensor.yaml for the camera what is T_BS?

# Sensor extrinsics wrt. the body-frame.
T_BS:
  cols: 4
  rows: 4
  data: [ 0.0356961,   0.0821689,   0.99597895, -0.01150323,
        -0.99798199, -0.04943879,  0.03984663, -0.01248458,
        0.05251415, -0.99539142,  0.08023831,  0.01387081,
        0.,          0.,          0.,          1.      ]

maym86, 2016-07-15:
Looks like it's the same?

aforster, 2016-07-15:
Not sure where you got this sensor.yaml from but in okvis you can find the coordinate system definition in the readme:

>Outputs and frames

>In terms of coordinate frames and notation,

>    W denotes the OKVIS World frame (z up),
>    C_i denotes the i-th camera frame
>    S denotes the IMU sensor frame
>    B denotes a (user-specified) body frame.

>The output of the okvis library is the pose T_WS as a position r_WS and quaternion q_WS, followed by the velocity in World frame v_W and gyro biases (b_g) as well as accelerometer biases (b_a). See the example application to get an idea on how to use the estimator and its outputs (callbacks returning states).

closed 2017-02-13 by maym86