ethz-asl/okvis #20: Error while "make"

HDaoud, 2016-08-25:
Hello,

Thanks for sharing the code, I tried to compile it but I'm getting this error:

>/usr/include/c++/5/fenv.h:58:11: error: ‘::fenv_t’ has not been declared
   using ::fenv_t;
           ^
/usr/include/c++/5/fenv.h:59:11: error: ‘::fexcept_t’ has not been declared
   using ::fexcept_t;
           ^
/usr/include/c++/5/fenv.h:62:11: error: ‘::feclearexcept’ has not been declared
   using ::feclearexcept;
           ^
/usr/include/c++/5/fenv.h:63:11: error: ‘::fegetexceptflag’ has not been declared
   using ::fegetexceptflag;
           ^
/usr/include/c++/5/fenv.h:64:11: error: ‘::feraiseexcept’ has not been declared
   using ::feraiseexcept;
           ^
/usr/include/c++/5/fenv.h:65:11: error: ‘::fesetexceptflag’ has not been declared
   using ::fesetexceptflag;
           ^
/usr/include/c++/5/fenv.h:66:11: error: ‘::fetestexcept’ has not been declared
   using ::fetestexcept;
           ^
/usr/include/c++/5/fenv.h:68:11: error: ‘::fegetround’ has not been declared
   using ::fegetround;
           ^
/usr/include/c++/5/fenv.h:69:11: error: ‘::fesetround’ has not been declared
   using ::fesetround;
           ^
/usr/include/c++/5/fenv.h:71:11: error: ‘::fegetenv’ has not been declared
   using ::fegetenv;
           ^
/usr/include/c++/5/fenv.h:72:11: error: ‘::feholdexcept’ has not been declared
   using ::feholdexcept;
           ^
/usr/include/c++/5/fenv.h:73:11: error: ‘::fesetenv’ has not been declared
   using ::fesetenv;
           ^
/usr/include/c++/5/fenv.h:74:11: error: ‘::feupdateenv’ has not been declared
   using ::feupdateenv;
           ^

HDaoud, 2016-09-08:
The problem is happening when compiling timing:

>In file included from /usr/include/boost/detail/fenv.hpp:69:0,
                 from /usr/include/boost/math/tools/config.hpp:291,
                 from /usr/include/boost/math/special_functions/sign.hpp:16,
                 from /usr/include/boost/lexical_cast.hpp:167,
                 from /usr/include/boost/date_time/string_parse_tree.hpp:13,
                 from /usr/include/boost/date_time/period_parser.hpp:14,
                 from /usr/include/boost/date_time/date_facet.hpp:23,
                 from /usr/include/boost/date_time/gregorian/gregorian_io.hpp:16,
                 from /usr/include/boost/date_time/gregorian/gregorian.hpp:31,
                 from /usr/include/boost/date_time/posix_time/time_formatters.hpp:12,
                 from /usr/include/boost/date_time/posix_time/posix_time.hpp:24,
                 **from /home/hdaoud/Documents/okvis/okvis_timing/include/okvis/timing/Timer.hpp:44,**
                **from /home/hdaoud/Documents/okvis/okvis_timing/src/Timer.cpp:34:**

aforster, 2016-09-08:
You could deactivate the timers, remove the Timer class and don't include the posix header as a workaround.

HDaoud, 2016-09-08:
Hi Andreas,

I tried to deactivate timers by adding the following:
`#ifndef DEACTIVATE_TIMERS`
`#define DEACTIVATE_TIMERS`

to the top of `Frontend.hpp`, and I removed the timer by following
`add_library(${PROJECT_NAME}
  **#src/Timer.cpp**
  src/NsecTimeUtilities.cpp
)`

I also commented the include of posix in timer.hpp

but I'm getting a lot of errors in frontend.cpp class :sweat: 

>/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:268:3: error: ‘matchStereoTimer’ was not declared in this scope
   matchStereoTimer.stop();
   ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp: At global scope:
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:274:6: error: ‘Frontend’ has not been declared
 bool Frontend::propagation(const okvis::ImuMeasurementDeque & imuMeasurements,
      ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:280:69: error: non-member function ‘bool okvis::propagation(const ImuMeasurementDeque&, const okvis::ImuParameters&, okvis::kinematics::Transformation&, okvis::SpeedAndBias&, const okvis::Time&, const okvis::Time&, Eigen::Matrix<double, 15, 15>*, Eigen::Matrix<double, 15, 15>*)’ cannot have cv-qualifier
                            Eigen::Matrix<double, 15, 15>* jacobian) const {
                                                                     ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:295:6: error: ‘Frontend’ has not been declared
 bool Frontend::doWeNeedANewKeyframe(
      ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp: In function ‘bool okvis::doWeNeedANewKeyframe(const okvis::Estimator&, std::shared_ptr<okvis::MultiFrame>)’:
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:304:8: error: ‘isInitialized_’ was not declared in this scope
   if (!isInitialized_)
        ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:364:17: error: ‘keyframeInsertionOverlapThreshold_’ was not declared in this scope
   if (overlap > keyframeInsertionOverlapThreshold_
                 ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:365:18: error: ‘keyframeInsertionMatchingRatioThreshold_’ was not declared in this scope
       && ratio > keyframeInsertionMatchingRatioThreshold_)
                  ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp: At global scope:
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:373:5: error: ‘Frontend’ has not been declared
 int Frontend::matchToKeyframes(okvis::Estimator& estimator,
     ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp: In function ‘int okvis::matchToKeyframes(okvis::Estimator&, const okvis::VioParameters&, uint64_t, bool&, bool, double*, bool)’:
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:398:44: error: ‘briskMatchingThreshold_’ was not declared in this scope
                                            briskMatchingThreshold_,
                                            ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:403:7: error: ‘matcher_’ was not declared in this scope
       matcher_->match<MATCHING_ALGORITHM>(matchingAlgorithm);
       ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:403:41: error: expected primary-expression before ‘>’ token
       matcher_->match<MATCHING_ALGORITHM>(matchingAlgorithm);
                                         ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:422:44: error: ‘briskMatchingThreshold_’ was not declared in this scope
                                            briskMatchingThreshold_,
                                            ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:427:7: error: ‘matcher_’ was not declared in this scope
       matcher_->match<MATCHING_ALGORITHM>(matchingAlgorithm);
       ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:427:41: error: expected primary-expression before ‘>’ token
       matcher_->match<MATCHING_ALGORITHM>(matchingAlgorithm);
                                         ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:434:27: error: ‘isInitialized_’ was not declared in this scope
     if (kfcounter == 0 && isInitialized_)
                           ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:436:73: error: there are no arguments to ‘runRansac3d2d’ that depend on a template parameter, so a declaration of ‘runRansac3d2d’ must be available [-fpermissive]
                     estimator.multiFrame(currentFrameId), removeOutliers);
                                                                         ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:436:73: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:440:10: error: ‘isInitialized_’ was not declared in this scope
[... 80 lines cut ...]
   descriptorExtractors_.clear();
   ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:820:26: error: ‘numCameras_’ was not declared in this scope
   for (size_t i = 0; i < numCameras_; ++i) {
                          ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:829:17: error: ‘briskDetectionThreshold_’ was not declared in this scope
                 briskDetectionThreshold_, briskDetectionOctaves_, 
                 ^
[... cut ...]
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:836:17: error: ‘briskDescriptionRotationInvariance_’ was not declared in this scope
                 briskDescriptionRotationInvariance_,
                 ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:837:17: error: ‘briskDescriptionScaleInvariance_’ was not declared in this scope
                 briskDescriptionScaleInvariance_)));
                 ^
/home/hdaoud/Documents/okvis/okvis_frontend/src/Frontend.cpp:839:18: error: ‘featureDetectorMutexes_’ was not declared in this scope
   for (auto it = featureDetectorMutexes_.begin();

aforster, 2016-09-08:
Seems like that broke it even more... 
Can you just comment out the `ifdef` etc such that only this line stays:

https://github.com/ethz-asl/okvis/blob/master/okvis_frontend/include/okvis/Frontend.hpp#L62

and

https://github.com/ethz-asl/okvis/blob/master/okvis_multisensor_processing/include/okvis/ThreadedKFVio.hpp#L95 

I must say I don't understand why you are getting this error. I have never encountered this issue on Ubuntu...

HDaoud, 2016-09-21:
Hi Andreas,

Thanks for your solution, I followed the recommended solution above, and I started to get errors regarding openCV, so I added the following to **CMakeLists.txt**:

>89 #OpenCV
90 find_package(OpenCV REQUIRED)

And it worked! :smile: I think you have to update the **CMakeLists.txt** to include **OpenCV**

[image]

closed 2016-09-21 by HDaoud