ethz-asl/okvis #23: Stereo only?

lexvandersluijs, 2016-11-15:
Hi,

Just a short question: do I understand correctly that OKVIS currently only supports stereo setups? Any plans to extend to monocular?

Thanks,
Lex

weblucas, 2016-11-16:
It already supports monocular. Your parameter file needs to contain calibration for only one camera.

xiaopenghuang, 2017-01-05:
@weblucas the formula is the same with the stereo or any difference,

wow2006, 2017-02-13:
Hello,
I want to use **stereo only** without IMU. **Can I use this library without IMU?!**
When I remove `okvis_estimator.addImuMeasurement(t_imu, acc, gyr);` at `okvis_app_synchronous.cpp` no update done to viewers.
Can you give me hint?!

aforster, 2017-02-13:
No. See #22

Mrzzhao, 2017-04-11:
@weblucas hello,I have run the programm successfully,and it looks beautiful. now I want to know whether it support the kinect2 ? thanks.