ethz-asl/okvis #24: okvis_ceres_test and okvis_multi_sensor_processing unit test failure

yonidishon, 2016-11-21:
Hello,
I'm getting unit-test errors and they are different depends on the build configuration Debug/Release.

Other than this fails the main application **okvis_app_synchronous** is building and executing fine and I can run it with no issues on the "Machine room" sequence (Euroc dataset). 

However I couldn't reproduce the results reported on "Vicon Room" ( IJRR 2014)  by "Keyframe-based visual–inertial odometry using nonlinear optimization" ICRA 2015.

Thank you,
Yonatan.
________________________________________________________________________________
#### Release unit-test output: ######
../okvis/build/Release/okvis_ceres$ ./okvis_ceres_test 
[==========] Running 6 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 6 tests from okvisTestSuite
[ RUN      ] okvisTestSuite.Estimator
case 0, 0
Frame 0 successfully added.
Frame 1 successfully added.
Frame 2 successfully added.
Frame 3 successfully added.
Frame 4 successfully added.
Frame 5 successfully added.
Frame 6 successfully added.
== TRY MARGINALIZATION ==
== LAST OPTIMIZATION ==
estimated T_WS: 
           1 -1.54699e-05  7.59601e-06   0.00182203
 1.54699e-05            1  1.39194e-06  -0.00206711
-7.59603e-06 -1.39182e-06            1   0.00018551
           0            0            0            1
correct T_WS: 
 1  0  0  0
 0  1  0 10
 0  0  1  0
 0  0  0  1
1.00045
unknown file: Failure
C++ exception with description "[Exception] 
####  ../okvis/okvis_ceres/test/TestEstimator.cpp:230: TestBody() assert((speedAndBias_est - speedAndBias).norm()< 0.04) failed: speed and biases not close enough" thrown in the test body. #### 
[  FAILED  ] okvisTestSuite.Estimator (1122 ms)
[ RUN      ] okvisTestSuite.HomogeneousPointError
run the solver... 
[       OK ] okvisTestSuite.HomogeneousPointError (1 ms)
[ RUN      ] okvisTestSuite.ReprojectionError
set up a random geometry...  [ OK ] 
set up a random camera geometry...  [ OK ] 
setting local parameterization for pose...  [ OK ] 
create N=100 visible points and add respective reprojection error terms...  [ OK ] 
run the solver... 
initial T_WS :  -0.921129  -0.329734  -0.206871    9.72266
 -0.324387   0.944003 -0.0602677   0.255295
  0.215159  0.0115921   -0.97651   -5.08655
         0          0          0          1
optimized T_WS :  -0.921631  -0.327145  -0.208737    9.57852
  -0.32134   0.944923 -0.0621359  -0.533125
  0.217568 0.00980934  -0.975996   -5.25324
         0          0          0          1
correct T_WS :  -0.921668  -0.327274  -0.208374    9.57751
 -0.321491   0.944879 -0.0620309  -0.533782
  0.217189 0.00981854   -0.97608   -5.24732
         0          0          0          1
[       OK ] okvisTestSuite.ReprojectionError (1 ms)
[ RUN      ] okvisTestSuite.ImuError
setting local parameterization for pose...  [ OK ] 
run the solver... 
initial T_WS_1 :  0.893578  0.157643  0.420319   1.33777
-0.118954  0.985993 -0.116913  0.745465
-0.432862 0.0544722  0.899813   1.48603
        0         0         0         1
[... 196 lines cut ...]
../okvis/build/Release/okvis_multisensor_processing$ ./okvis_multisensor_processing_test 
[==========] Running 11 tests from 2 test cases.
[----------] Global test environment set-up.
[----------] 6 tests from FrameSynchronizerTest
[ RUN      ] FrameSynchronizerTest.ConstructDestruct
[       OK ] FrameSynchronizerTest.ConstructDestruct (379 ms)
[ RUN      ] FrameSynchronizerTest.CorrectOrder
[       OK ] FrameSynchronizerTest.CorrectOrder (372 ms)
[ RUN      ] FrameSynchronizerTest.OneMissing
E1121 10:27:37.920555 21633 FrameSynchronizer.cpp:104] Dropping frame with id 14
[       OK ] FrameSynchronizerTest.OneMissing (376 ms)
[ RUN      ] FrameSynchronizerTest.TwoMissing
E1121 10:27:38.296674 21633 FrameSynchronizer.cpp:104] Dropping frame with id 24
E1121 10:27:38.296736 21633 FrameSynchronizer.cpp:104] Dropping frame with id 26
[       OK ] FrameSynchronizerTest.TwoMissing (373 ms)
[ RUN      ] FrameSynchronizerTest.OneDouble
E1121 10:27:38.669687 21633 FrameSynchronizer.cpp:104] Dropping frame with id 34
[       OK ] FrameSynchronizerTest.OneDouble (373 ms)
[ RUN      ] FrameSynchronizerTest.OneOutOfOrder
[       OK ] FrameSynchronizerTest.OneOutOfOrder (394 ms)
[----------] 6 tests from FrameSynchronizerTest (2267 ms total)

[----------] 5 tests from OkvisVioInterfaces
[ RUN      ] OkvisVioInterfaces.testConstructionDestruction
####  Floating point exception (core dumped) ####

aforster, 2016-11-21:
can you try to apply #16 ?

yonidishon, 2016-11-22:
Thank you very much.
I applied #16 and indeed it fixed the 1st problem (EstimatorTest.cpp). sorry for not noticing this pull request.
**However it didn't fix the failure on  OkvisVioInterfaces.testConstructionDestruction**, 

I change some values around to try and help it but it is going deeper and deeper.
_________________________________________________________________________________
for this I had to  first change the line 12 in testDataGenerator.cpp:
`class TestDataGenerator {
 public:
  static const okvis::cameras::NCameraSystem getTestCameraSystem(int num_cams) {
    okvis::cameras::NCameraSystem nCameraSystem;
    for (int i = 0; i < num_cams; ++i) {
      Eigen::Matrix4d T_SC_e;
      T_SC_e << 1, 0, 0, i*0.01, // TODO(gohlp): does that make sense?
                0, 1, 0, 0,
                0, 0, 1, 0,
                0, 0, 0, 1; // << FAILED TEST: changed from 0 -> 1 in last element of matrix`
To remove the first Floating point exception..

Then I got another on because the constructor of ThreadedKFVio class is dividing by zero when parameters.sensors_information.cameraRate is zero in this test (no value in initialization so set to zero by my compiler)

aforster, 2016-11-22:
You fixed everything then? I guess the tests aren't very well tested ;) But I am sure Stefan might appreciate if you can add your other fixes into a PR. At least your fixes make sense to me.

yonidishon, 2016-11-23:
Not quite yet.

The reason I went to verified the unit-test in the first place is because I failed to repeat the results from the paper on the VICON LOOP dataset (14 min loops in the vicon room).
The .yaml files supplied with the dataset are a bit strange (not valid .yaml files and missing some parameters : sigma_ba, sigma_bg and makKeyPointNum). 
I'm still ivestigating why my run results aren't close to the .csv recorded estimates that are come with the dataset - I have a small noise jitter in the trajectory.
Few directions I got:

1.The outliers from the keypoint tracking are causing to some noise in the estimation  -  in the paper (and dataset documentation) the authors wrote that they fed the system with the the same tracked points, but it seems that in the source these functions aren't yet implemented. 
2. I took the missing parameters from the config file supplied with the code (and that you cited in pull request #16) but it seems that they don't reproduce the results. 

So I'm trying to for the online estimation of the noise parameter to plug them after one run as the priors for the next run and hopefully it will then generate the reported results?

So currently it does seem that the unit-test aren't verified,but I esitate to say so due to the high-quality of the code and documentation so I don't want to offer a solution that is not verified (If I knew the code is really work as expected I would have a much easier time on doing so).

I can offer a partial pull request but I prefer to be sure.
Would you have the set of parameters to reconstruct the Vicon 14min results?

aforster, 2016-12-01:
I can't help you with that. Maybe it is better to check with S. Leutenegger directly via mail?

yonidishon, 2016-12-04:
Thanks Andreas,
I already tried to contact him directly (e-mail) two weeks ago with a ping a week ago about this issue with no response so far.

I thought you might have an insight since you work in the same laboratory, no?
Thank you for your help anyway.