ethz-asl/okvis #25: How to convert kalibr calibration data(imu) to okvis config file?

maoLB, 2016-12-21:
I have assembled a camera-imu device. First i use kalibr calibration method to calibrate my own device.
The calibration yaml look like these:
imu0:
  T_i_b:
  - [1.0, 0.0, 0.0, 0.0]
  - [0.0, 1.0, 0.0, 0.0]
  - [0.0, 0.0, 1.0, 0.0]
  - [0.0, 0.0, 0.0, 1.0]
  accelerometer_noise_density: 0.001409
  accelerometer_random_walk: 5.009e-05
  gyroscope_noise_density: 3.894e-05
  gyroscope_random_walk: 6.899e-08
  model: calibrated
  rostopic: /imu
  time_offset: 0.0
  update_rate: 60.0

I'm confuse to generate my own config file when compare my yaml to your config_fpga_p2_euroc.yaml. 
Just like above, i get four params finally but the config yaml has eight params.
    (a_max: 176.0 # acceleration saturation [m/s^2]
    g_max: 7.8 # gyro saturation [rad/s]
    sigma_g_c: 12.0e-4 # gyro noise density [rad/s/sqrt(Hz)]
    sigma_a_c: 8.0e-3 # accelerometer noise density [m/s^2/sqrt(Hz)]
    sigma_bg: 0.03 # gyro bias prior [rad/s]
    sigma_ba: 0.1 # accelerometer bias prior [m/s^2]
    sigma_gw_c: 4.0e-6 # gyro drift noise density [rad/s^s/sqrt(Hz)]
    sigma_aw_c: 4.0e-5 # accelerometer drift noise density [m/s^2/sqrt(Hz)])
I study camera-imu for not a long time. Can you help me?

weblucas, 2017-01-05:
there are conversion scripts on the Kalibr in aslam_offline_calibration/kalibr/python/exporters

paulhaile, 2017-04-25:
Pretty simple follow up question. T_sc is the transformation from the imu to the camera if I'm not mistaken. The module that we are using defines its coordinate system along the same frame of the camera. (z axis going forward through the lens, y axis straight downwards and x axis to the right). In this representation is our T_SC parameter in our config file simply the identity matrix? We have been attempting this and not getting any nice results. Any help is appreciated. 
Thank you
[image]

VincentFEI, 2017-05-20:
you can use comman:
“ kalibr_aslam_config --cam kalibr_workspace/src/calibrate/camera-imu-loitor/camchain-imucam-dynamic.yaml --mav mav0”
you should put you camchain yaml and you will get your okvis yaml.

weblucas, 2017-05-25:
wher is the IMU? T_SC is the other way around point in the sensor frame = T_SC * point in the camera frame

Aishkrish18, 2023-08-04:
can you tell me what is the command to convert to the config file format? When i try it out i get this error: 
[image]

VincentFEI, 2023-08-04:
您好，收到您的邮件啦！