ethz-asl/okvis #26: Segmentation fault on Ubuntu 16.04

yanhangpublic, 2016-12-26:
Hi,
I'm now trying the okvis on ubuntu 16.04. The compilation went smoothly, but I can not make it run even on the example data.

Here is my environment (everything is pretty much up-to-date):
Ubuntu 16.04, OpenCV 3.2, Nvidia-driver 367,  Cuda 8.0, Eigen 3.3~beta1-2, Ceres 1.12

Here is my problem: when I tried to run `okvis_app_synchronous` on example data (MH_01_easy, MH_04_difficulty), there will be a segmentation fault error immediately after the viewer windows show up. When I tried to run it in debugging mode,  the program breaks at the following line inside Eigen source:
```
template<int StoreMode, int LoadMode, typename PacketType>
EIGEN_DEVICE_FUNC EIGEN_STRONG_INLINE void assignPacket(Index row, Index col)
{
    m_functor.template assignPacket<StoreMode>(&m_dst.coeffRef(row,col), m_src.template packet<LoadMode,PacketType>(row,col));
}
```
Although this might not be useful...

I also tried some 'printf' debugging (print some text before and after a line to see where the program actually breaks). It seems that `if (matchedFrames_.PopBlocking(&frame_pairs) == false)` at line 729 inside ThreadedKFVio.cpp is suspicious. Maybe the `pthread` is not compatible with the memory of newer Eigen? I'm not confidence about this.

Does someone has the same issue? The problem seems to be related with Eigen. Which version of Eigen3 should I use?

Thanks!

higerra, 2016-12-26:
I finally get it run. The problem seems to be Ceres solver. Previously I was using Ceres 1.12 on my system. After I switched to the internal Ceres, it works!

I noticed that the internal Ceres used is version 1.9. Maybe it worth figuring out why it fails on 1.12?

closed 2016-12-26 by yanhangpublic

Jackzhemliu, 2017-09-06:
how did you switch to the internal Ceres?

chanzw, 2019-05-16:
Hello,when I run the okvis_app_synchronous on example data (MH_01_easy),it doesn't work ,and it show me the message like this :
W0516 20:56:31.486670 15535 Frontend.cpp:282] - Skipping propagation as only one IMU measurement has been given to frontend. Normal when starting up.
Segmentation fault (core dumped)

can u tell me the reason? thank you!