ethz-asl/okvis #27: Assumptions about gravity direction and initial pose?

mklingen, 2017-01-26:
I have gotten okvis to successfully run on the MAV datasets, and have started feeding my own data into the algorithm. In general it just drifts like crazy, which is not surprising considering the many things that could be wrong (calibration of IMU to camera, time synchronization, etc.). While I was looking through the configs and code I was wondering: how do you estimate the acceleration due to gravity? Is there any assumption about the starting direction of gravity w.r.t the IMU? Are you assuming that gravity is subtracted from the IMU measurements before they are fed into the algorithm, or are you estimating that value online?

aforster, 2017-01-27:
Check the addStates function in the estimator class: https://github.com/ethz-asl/okvis/blob/master/okvis_ceres/src/Estimator.cpp#L110

At the beginning `initPoseFromImu` is called https://github.com/ethz-asl/okvis/blob/master/okvis_ceres/src/Estimator.cpp#L811

This is where the initial pose in the world frame is set. If I remember right it checks the highest acceleration direction and sets roll&pitch to make it consistent.. Later refinements come implicitly with the tightly coupled approach. The only assumption is that the first poses are without other accelerations other than gravity.

mklingen, 2017-01-27:
Okay thank you, that helps me track down bugs (right now my best guess is incorrect calibration). I do get this however:

`[Frontend.cpp:761] Initializing pose from 2D-2D RANSAC: orientation only`

I assume that's normal even if the pose gets initialized from IMU? Would there be any problem with the dataset starting perfectly stationary (its mounted on a ground robot)?

mklingen, 2017-01-27:
I am also now starting to wonder about the applicability of this approach to data from a walking robot (which routinely experiences shocks while moving) as opposed to a MAV robot. Looking at the IMU data, we experience shocks of about +/- 20 m/s^2 every second or so.

mklingen, 2017-02-01:
FYI, I discovered that most of my issues were coming from an incorrect extrinsic calibration (I had T_SC as the inverse of what it should have been).

closed 2017-02-01 by mklingen