ethz-asl/okvis #28: windows build?

antithing, 2017-01-28:
Hi, thank you for this code. Will it be possible to build it on Windows?

fvzaur, 2017-02-17:
Its possible to build it for windows. OKVIS has no ROS dependency (non-ROS version, that is this version). I have built it for windows, it works fine.

zz1120122573, 2018-01-13:
Dear @zziz 
I'm struggling to build okvis on windows. I don't know about cmake, and I cannot find references on Google. Could you give me some suggestion or tutorial? Thanks a lot.

zziz, 2018-01-13:
Hi @zz1120122573 
First you should build all libraries that OKVIS depends on. README page has the list of dependencies. And finally link all these libraries to OKVIS build as a final step. How to use cmake? Here is the steps from http://assimp.sourceforge.net/lib_html/cmake_build.html (check out the website as well). 

- Download the latest CMake from http://www.cmake.org/ and install it
- Run CMake GUI (at the time of this writing, its called cmake-gui.exe on Windows and appears in the start menu after installation)
- Point it to the location of the assimp source code. The UI should now look like the screenshot above
- Also point it to the folder where you want the binaries to be placed in.
- Lines marked in red mark settings you haven't confirmed yet, so check if everything is fine and press "Configure" again (if you rebuild assimp later, only changed or newly added settings will appear in red since the last build settings are cached by CMake!)
- Press "Generate" and pick a suitable output build system
- Open the generated solution/project files and have fun --> Go to this folder and Build the project! Done!
Do this configure-generate-build steps for each dependent library and OKVIS at last. 
I should warn you that it can be tedious task to make it work on windows compared to linux. If windows is not absolutely critical part, then I recommend ubuntu 14.04. But anyways, it does work on windows. Use Visual Studio 2013.

zz1120122573, 2018-01-15:
Dear @zziz .
Thanks for your reply! I have build all dependences in README, but there are three library( BRISK, OPENGV and CERES) not in README. They seems to be download and built during build OKVIS. I try to build them outside, but the OKVIS may download the three libraris everytime.
The BREASK use some C++11 code, if I use VS2013, I have to change the source code. Did you try VS2015 and failed? Thanks again.

zziz, 2018-01-15:
@zz1120122573 
I think automatic download and build doesn't work on windows. What you should do is configure cmakefile for OKVIS so that it uses local build of brisk, opengv and ceres. Variable name for brisk is USE_SYSTEM_BRISK, and for ceres is USE_SYSTEM_CERES. Check the checkbox and set ceres_DIR to your local build (similarly for others). I have built it for VS2013 and VS2017. Works in both cases.  Haven't checked it on VS2015. My guess is it will work.

zz1120122573, 2018-01-16:
Dear @zziz 
I have compiled opengv by VS2015, but how can I include opengv in OKVIS? By cmakelist or setting in VS properties? Shall I delete opengv_external in OKVIS? Thanks for your support.

zziz, 2018-01-17:
Dear @zz1120122573 
From CMake Add Entry add OpenGV_LIBRARIES variable and set type to PATH and value to your directory of opengv.lib. If opengv_external still shows up in your project built, then just delete it and start building sub-projects of okvis one by one and link necessary libraries as you go. Note that you only need to link headers for all subprojects other than applications. Link libraries in application sub-project.

zz1120122573, 2018-01-17:
Dear @zziz 
Thank you very much, I finally built OKVIS on windows. Thanks for your patiece.

zziz, 2018-01-17:
Dear @zz1120122573 
It was my pleasure. Glad to hear that!

antithing, 2018-07-24:
@zz1120122573  Sorry to bug you, but can you tell me how you got this to compile? I am having trouble with pthreads.h missing. Did you re-write the threadsafe que? Or find another workaround?

478161648, 2018-11-20:
did you install the visensor？There is a problem ,“ Unknown CMake command "catkin_package".”. do you know how to slove it? thank you.

478161648, 2018-11-20:
can you tell me how to install the brisk? there is a problem "Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)", should I install the pythonLibs? thank you.

478161648, 2018-11-20:
I can't compile the brisk...

weblucas, 2019-01-07:
hi mirella, open the cmake-gui and make sure that all dependencies were found.

mirellameelo, 2019-01-14:
@weblucas I was missing Ceres Solver dependency... Tks!
 As @zz1120122573 suggested, I compiled the projects one by one, and I already have the libraries,  but the executable (okvis_app_synchronous) is having linking issues with okvis_ceres.lib, okvis_frontend.lib, and okvis_multisensor_processing.lib. The paths are all right in the properties... No idea what's going on...

Other question, Boost dependency must be built in "static" or "dynamic" configuration? 

Tks!

mirellameelo, 2019-01-16:
@weblucas I was missing Ceres Solver dependency... Tks!

As @zz1120122573 suggested, I compiled the projects one by one, and I already have the libraries, including the external libs ceres.lib and brisk.lib, but I'm having problems with opengv. I'm using CMake to generate the .sln, and I have it, but the compilation is never ending!