ethz-asl/okvis #31: How to save position data as a txt file with timestamp?

zxw1992, 2017-03-21:


weblucas, 2017-03-21:
you have to write some code in the function publishFullStateAsCallback in the file [okvis_app_synchronous.cpp](https://github.com/ethz-asl/okvis/blob/master/okvis_apps/src/okvis_app_synchronous.cpp)

zxw1992, 2017-03-22:
I got the idea that I can save position data and velocity data if I write some code at here:

`  std::stringstream postext;
    postext << "position = [" << r[0] << ", " << r[1] << ", " << r[2] << "]";
    cv::putText(_image, postext.str(), cv::Point(15,15),
                cv::FONT_HERSHEY_COMPLEX, 0.5, cv::Scalar(255,255,255), 1);
    std::stringstream veltext;
    veltext << "velocity = [" << speedAndBiases[0] << ", " << speedAndBiases[1] << ", " << speedAndBiases[2] << "]";
    cv::putText(_image, veltext.str(), cv::Point(15,35),
                    cv::FONT_HERSHEY_COMPLEX, 0.5, cv::Scalar(255,255,255), 1);`

But, as I want to compare the estimated results with the groundtruth, it seems necessary to save the timestamp simultaneously. However, I have no idea about how to do it. Can you give me some hints? 
Thank you very much.

Mrzzhao, 2017-04-17:
@zxw1992 hello，I want to save the pose to txt ,are you successful? could you tell me how to operate?

447425299, 2017-04-24:
@zxw1992 I want to do the same things ,  Only the timestamp i can not find, are you succeed?

447425299, 2017-04-24:
Thanks for your sharing, but i can hardly find timestamp to output

Mrzzhao, 2017-04-24:
@447425299  I can output the timestamp ,but it cannot Synchronize，so,I  save the position and timestamp respectively,then I combine them together into .csv.What`s more,I compare it with the groundtruth,and get the result.

447425299, 2017-04-24:
@zxw1992 TUM dataset has the Script to do it.
https://github.com/tum-vision/mono_dataset_code　The .m files

447425299, 2017-04-24:
@zxw1992 TUM dataset has the Script to do it.
https://github.com/tum-vision/mono_dataset_code