ethz-asl/okvis #34: Crashed with Ubuntu 16.04

Userskii, 2017-05-10:
I tried the code with 16.04 but it just crashed with the following info:

SUMMARY
========

PARAMETERS
 * /okvis_node/config_filename: /home/00/cat...
 * /okvis_node/mesh_file: firefly.dae
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    okvis_node (okvis_ros/okvis_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[okvis_node-1]: started with pid [6795]
I0510 16:46:19.477212  6795 VioParametersReader.cpp:85] Opened configuration file: /home/00/catkin_ws/src/okvis_ros/okvis/config/config_fpga_p2_euroc.yaml
I0510 16:46:19.477421  6795 VioParametersReader.cpp:173] imageDelay=0
I0510 16:46:19.477526  6795 VioParametersReader.cpp:275] Custom World frame provided T_Wc_W=
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
I0510 16:46:19.477552  6795 VioParametersReader.cpp:554] Found calibration in configuration file for camera 0
I0510 16:46:19.477562  6795 VioParametersReader.cpp:554] Found calibration in configuration file for camera 1
I0510 16:46:19.485872  6795 VioParametersReader.cpp:363] Radial tangential pinhole camera 0 with T_SC=
 0.0148655  -0.999881  0.0041403 -0.0216401
  0.999557  0.0149672  0.0257155  -0.064677
-0.0257744 0.00375619   0.999661 0.00981073
         0          0          0          1
I0510 16:46:19.809892  6795 VioParametersReader.cpp:363] Radial tangential pinhole camera 1 with T_SC=
 0.0125553  -0.999755  0.0182238 -0.0198436
  0.999599  0.0130119  0.0251588  0.0453689
-0.0253898  0.0179006   0.999517 0.00786212
         0          0          0          1
I0510 16:46:19.809929  6795 VioParametersReader.cpp:412] IMU with transformation T_BS=
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
init done
W0510 16:47:33.659103  6827 Frontend.cpp:282] - Skipping propagation as only one IMU measurement has been given to frontend. Normal when starting up.
*** stack smashing detected ***: /home/00/catkin_ws/devel/lib/okvis_ros/okvis_node terminated
[okvis_node-1] process has died [pid 6795, exit code -6, cmd /home/00/catkin_ws/devel/lib/okvis_ros/okvis_node /camera0:=/cam0/image_raw /camera1:=/cam1/image_raw /calibration0:=/cam0/calibration /calibration1:=/cam1/calibration /imu:=/imu0 __name:=okvis_node __log:=/home/00/.ros/log/0a279138-3555-11e7-ba23-9c5c8ec1f302/okvis_node-1.log].
log file: /home/00/.ros/log/0a279138-3555-11e7-ba23-9c5c8ec1f302/okvis_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Any suggestions?

weblucas, 2017-05-16:
The info that you show is ok. It is what everybody gets. The error itself is not informative at all. I never tried on kinect, but I would focus on the shared libraries. such as ceres, eigen and opencv.

Best

Lucas

Userskii, 2017-05-18:
@weblucas Thanks for your comment. Yes you are right. The thread will have an issue on my machine when using ceres to optimize the problem. To be exact, it occurs at "mapPtr_->solve();" in the source file of "Estimator.cpp". I am looking into the problem to see why I get this problem.

Userskii, 2017-05-19:
@weblucas  I found this issue useful https://github.com/googlecartographer/cartographer_ros/issues/14
This is caused by libsuitesparse-dev which is 4.4.6 on xenial.

closed 2017-05-19 by Userskii

salahkhan94, 2018-06-08:
@MerlotQ  were you able to install OKVIS on Ubuntu 16.04 ?

ferreram, 2018-08-27:
Hi @MerlotQ ,
Would you mind detailing how you proceed to make it work and fix the *stack smashing detected* issue?
Thanks in advance!