ethz-asl/okvis #35: Start fail using own sensor.

cheneyfan, 2017-05-16:
F0516 16:45:42.349895  5751 parameter_block.h:164] Check failed: new_parameterization->GlobalSize() == size_ Invalid parameterization for parameter block. The parameter block has size 7 while the parameterization has a global size of 6. Did you accidentally use the wrong parameter block or parameterization?

weblucas, 2017-05-16:
Poses are described as 3d point and a quaternion. 7 elements. What are you trying to do?

cheneyfan, 2017-05-17:
Does it in sync？
                             Time
IMU data               1494489055 622111111
IMU data               1494489055 722111111
IMU data               1494489055 922111111
Camera 0  data      1494489055 822111111 
IMu data                1494489056 022111111
Camera 1  data      1494489055 822111111

cheneyfan, 2017-05-17:
I only add okvis_estimator.addImuMeasurement（...），it runs ok but large drift.
But when i add okvis_estimator.addImage（...）at the same time， the problems F0516 occurs.

While, i record camera and imu data with the same time sequence and run with default demo, it runs ok and drifts slightly. So how can i run okvis real-time?

weblucas, 2017-05-17:
you are adding the event out of order
IMU data 1494489055 **7**22111111
IMU data 1494489055 **9**22111111
Camera 0 data 1494489055 **8**22111111

on the offline demo, it syncs to you.

cheneyfan, 2017-05-18:
your code is 
do{
{
while（t_imu<=t）

So， the time sequence of IMU will have one data is older than Camera