ethz-asl/okvis #36: How to compare With EuRoc groundtruth?

David-Willo, 2017-05-17:
Now I am running Okvis using EuRoc MAV Dataset and trying to compare the pose with the ground truth given in the dataset. I add some code in the publishFullStateAsCallback to  save position data and velocity data which are the same with the data displayed on the poseViewer. However, when i compare with the ground truth, i found the pose is different. How can i make a comparison ?

weblucas, 2017-05-17:
depending of the euroc sequence, the ground truth is provided in a different coordinate system, usually in the top of the UAV. OKVIS by default will provide the pose centered in the camera. you can change the config file to use a custom world frame. There you can add the tranformation between the imu(B-body) and the origin of the Euroc UAV (S) [look this image- you need T_SB](http://projects.asl.ethz.ch/datasets/lib/exe/fetch.php?w=400&media=platform.jpg)

David-Willo, 2017-05-18:
Thanks! Let me have a try.

aforster, 2017-05-18:
@weblucas default pose is IMU pose not camera if I remember correctly

David-Willo, 2017-05-18:
[image]
[image]
I change the world frame like this, but the output doesn't seem to change. Anything wrong?

David-Willo, 2017-05-18:
[image]
The origin world frame looks like this.

closed 2017-06-28 by David-Willo

Chandrahas1991, 2017-08-23:
@weblucas  Could you please confirm if the pose is IMU or camera centered ? README says IMU

Chandrahas1991, 2017-08-24:
@David-Willo How did you compute the extrinsics between okvis (IMU) output and ground truth ?

David-Willo, 2017-08-24:
@Chandrahas1991 Well, I made a mistake at that time.
To compare the OKVIS pose output with the EuRoc ground truth, we need to associate the data sequences (according to the timestamp), since the sequence length is usually different. Then we can compute the translation and rotation between the two coordinate system. Have a look at the TUM toolkit, it helps a lot.

Chandrahas1991, 2017-08-24:
@David-Willo Thanks for your response. Could you please share the link of that toolkit ?

Chandrahas1991, 2017-08-24:
@David-Willo Also, is it incorrect if for every datapoint's timestamp in the okvis estimate, i look for the closest timestamp in the groundtruth data ? (and then perform the transformation between the two coordinate systems)

David-Willo, 2017-08-28:
@Chandrahas1991 https://vision.in.tum.de/data/datasets/rgbd-dataset/tools there are some description and a download link in it.

Chandrahas1991, 2017-08-28:
@David-Willo Thanks!

jzubizarreta, 2017-09-26:
Same problem here! Did you manage to solve the problem?
If yes, could you share with me your tool to synchronize images with the ground truth pose of the cameras?

Thanks in advance,

Jon

Chandrahas1991, 2017-09-26:
@jzubizarreta Have a look at the link @David-Willo has shared. Specifically the "Absolute Trajectory Error" script. As long as your timestamps from both the systems are in same precision, the script will align the pose and ground-truth using a stable method. Additionally you can also save the aligned poses using the "-save_associations" flag.

jzubizarreta, 2017-09-30:
@Chandrahas1991 Thanks!

cumtchenchang, 2018-02-03:
@David-Willo same question. I change the cold, but I can not get the generate pose to compare the ground truth. May you share your cold, I just want you get OKVIS trajectory to evelate. Please.