ethz-asl/okvis #38: Calibration problem？

cheneyfan, 2017-05-24:
in the Camera-IMU calibration，i get “T_cam_imu”， then put it in T_SC in the config file directly？

T_cam_imu and T_SC both mean transformation from Sensor to Camera？

weblucas, 2017-05-24:
sensor = imu
pose_insensorframe= T_SC * pose_incameraframe 
vs
pose_incameraframe= T_cam_imu * pose_insensorframe

Are you using kalibr? you can use the exporter script for OKVIS
https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/exporters/kalibr_aslam_config

cheneyfan, 2017-05-25:
Yes, i’am using Kalibr.
Directly replace T_SC with T_cam_imu？ If not， how to transform T_cam_imu into T_SC？

weblucas, 2017-05-25:
No you have to invert the matrix. use this python script that it already prepare all the matrix to you
https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/exporters/kalibr_aslam_config

cheneyfan, 2017-05-26:
“T_cam_imu” in camchain-imucam-%BAGNAME%.yaml ， than convert it into “T_SC” in config_fpga_p2_euroc.yaml ？ 
can i just inverse “T_cam_imu”？

aforster, 2017-05-26:
Yes. Invert it.
`S` stands for Sensor which refers to the IMU. Please read the Readme.

Also check [this](http://paulfurgale.info/news/2014/6/9/representing-robot-pose-the-good-the-bad-and-the-ugly) for a good introduction to camera frame notation.

weblucas, 2017-05-26:
yes, but to avoid numerical errors. do not invert the 4x4 matrix. use this
approach
http://www.cg.info.hiroshima-cu.ac.jp/~miyazaki/knowledge/image/eq53.gif,

cheneyfan, 2017-05-27:
OK, I have finished the calibration.
The pose keeps unchanged when power on and it can run tracking normally within small and slight motion， but it will drift a lot with normal motion. How to optimize it？

weblucas, 2017-05-27:
What is the range of the reprojection error of your calibration?

cheneyfan, 2017-05-27:
Calibration results
===================
Reprojection error squarred (cam0):  mean 0.155326923144, median 0.0510172933023, std: 0.624074143921
Reprojection error squarred (cam1):  mean 0.15045563157, median 0.048621853134, std: 0.680495108651
Gyro error squarred (imu0):          mean 198.677348797, median 7.02152733236, std: 749.008430121
Accelerometer error squarred (imu0): mean 5.01081140157, median 0.276420872561, std: 18.4145340448

Transformation (cam0):
-----------------------
T_ci:  (imu to cam0): [m]
[[ 0.0364117  -0.99857832 -0.03892978 -0.00774579]
 [-0.99917728 -0.03707442  0.01643904 -0.00258252]
 [-0.01785897  0.03829918 -0.99910672  0.00072436]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam0 to imu): [m]
[[ 0.0364117  -0.99917728 -0.01785897 -0.00228543]
 [-0.99857832 -0.03707442  0.03829918 -0.00785827]
 [-0.03892978  0.01643904 -0.99910672  0.00046463]
 [ 0.          0.          0.          1.        ]]

timeshift cam0 to imu0: [s] (t_imu = t_cam + shift)
0.0

Transformation (cam1):
-----------------------
T_ci:  (imu to cam1): [m]
[[ 0.02330732 -0.99947719 -0.02240795 -0.10869984]
 [-0.9995838  -0.02367924  0.01647798 -0.0027767 ]
 [-0.01699996  0.02201457 -0.99961311 -0.00136361]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam1 to imu): [m]
[[ 0.02330732 -0.9995838  -0.01699996 -0.00026522]
 [-0.99947719 -0.02367924  0.02201457 -0.10867874]
 [-0.02240795  0.01647798 -0.99961311 -0.00375307]
 [ 0.          0.          0.          1.        ]]

timeshift cam1 to imu0: [s] (t_imu = t_cam + shift)
0.0

Baselines:
----------
Baseline (cam0 to cam1): [m]
[[ 0.99977725  0.01339852 -0.01630746 -0.10090936]
 [-0.01339245  0.9999102   0.00048138 -0.00029849]
 [ 0.01631245 -0.00026288  0.99986691 -0.0019622 ]
 [ 0.          0.          0.          1.        ]]
baseline norm:  0.100928875648 [m]

Gravity vector in target coords: : [m/s^2]
[ 5.27157227 -4.73801359 -6.7821717 ]

Calibration configuration
=========================

cam0
-----
  Camera model: pinhole
  Focal length: [419.7210910977609, 492.82337740835027]
  Principal point: [319.52734052614846, 235.52733517897832]
  Distortion model: equidistant
  Distortion coefficients: [-0.17748121501791742, 0.10571210592569351, -0.12395922727304214, 0.05614521113566777]
  Type: aprilgrid
  Tags:
    Rows: 6
    Cols: 6
    Size: 0.088 [m]
    Spacing 0.0264 [m]

cam1
-----
  Camera model: pinhole
  Focal length: [418.0268345400491, 490.8384144277561]
  Principal point: [317.8793551858768, 231.18805105856094]
  Distortion model: equidistant
  Distortion coefficients: [-0.16485391420830264, 0.06374731242574906, -0.06152118701051446, 0.024577910349410686]
  Type: aprilgrid
  Tags:
    Rows: 6
    Cols: 6
    Size: 0.088 [m]
    Spacing 0.0264 [m]

IMU configuration
=================

  Update rate: 200.0
  Accelerometer:
    Noise density: 0.0025802
    Noise density (discrete): 0.0364895383364
    Random walk: 0.00048889
  Gyroscope:
    Noise density: 0.00012901
    Noise density (discrete): 0.00182447691682
    Random walk: 6.23e-05

lusj, 2017-06-16:
@cheneyfan  have you got the propre value of "T_SC"?