ethz-asl/okvis #41: inconsistent equation in ImuError::propagation compared with ijrr15 paper equation 15

extr15, 2017-07-14:
hi @sleutenegger , seems the imu propagation model is not the same with the paper?
in function `ImuError::propagation`
```
      // Fd = I + Fc*dt
      Eigen::Matrix<double,15,15> F_delta = Eigen::Matrix<double,15,15>::Identity();
      // transform
      // delta pos / delta alpha
      F_delta.block<3,3>(0,3) = -okvis::kinematics::crossMx(acc_integral*dt + 0.25*(C + C_1)*acc_S_true*dt*dt);
      // delta pos / delta vel
      F_delta.block<3,3>(0,6) = Eigen::Matrix3d::Identity()*dt;
      // delta pos / delta bias gyr
      F_delta.block<3,3>(0,9) = dt*dv_db_g + 0.25*dt*dt*(C*acc_S_x*cross + C_1*acc_S_x*cross_1);
      // delta pos / delta bias acc
      F_delta.block<3,3>(0,12) = -C_integral*dt + 0.25*(C + C_1)*dt*dt;
```
you set the all the columns of the first three rows of `F_delta`, but in your ijrr15 paper, there are many zeros blocks in the first three rows in equation 15:

[image]

I also read your `RSS2013` paper, but same thing here. So could you please explain this? thanks for your time very much!

JingeTu, 2017-11-24:
Hi! Have you understand this `F_delta` now?
I know `F_delta` is a state transformation matrix, those identity matrix should be there.
I referred to the article `Quaternion kinematics for the error-state Kalman filter`. I have spent three days here. I know there are somethings called true state, nominal state, error state. Use error state to propagate covariance. However, I can not understand this either. I doubt these code referred to different formulations.

The article link is here http://www.iri.upc.edu/people/jsola/JoanSola/objectes/notes/kinematics.pdf. Please read the section 5.

JingeTu, 2017-11-24:
Hi! Have you understood this `F_delta`?
I know `F_delta` is a state transformation matrix, those identity matrices should be there.
I referred to the article `Quaternion kinematics for the error-state Kalman filter`. I have spent three days here. I know there are somethings called true state, nominal state, error state. Use error state to propagate covariances. However, I can not understand this either. I doubt these code referred to different formulations.

The article link is here http://www.iri.upc.edu/people/jsola/JoanSola/objectes/notes/kinematics.pdf. Please read the section 5.

extr15, 2017-12-03:
@JingeTu , Thanks, that's a good reference.
I am not very clear yet, but I think maybe the paper use euler integration while the code use midpoint integration. 
You may ref this: https://www.zhihu.com/question/64381223

LeoGo2017, 2018-06-21:
@extr15   I have the same question as you？ Do you understand it now？

extr15, 2018-06-29:
@LeoGo2017 No, I don't think it is right, maybe it's not that important, you can try with different version.

FanaticsKang, 2018-09-18:
@JingeTu Focus on the Frame of input and output. I think dv_db_g is added and other difference is come from the definition of Frame.

cggos, 2019-01-12:
@JingeTu Have you understood the `F_delta`? 
I have read the `ESKF` part of the book *Quaternion kinematics for the error-state Kalman filter*, but I could not understand it in okvis yet.

1ADZX, 2023-11-29:
@extr15 hi, I saw you asked about F_delta, so I would like to ask what is the theory related to the following code?

_dalpha_db_g += dt * C_1;//，
    const Eigen::Matrix3d cross_1 =dalpha_db_g
        dq.inverse().toRotationMatrix() * cross + okvis::kinematics::rightJacobian(omega_S_true * dt) * dt;
    const Eigen::Matrix3d acc_S_x = okvis::kinematics::crossMx(acc_S_true);
    Eigen::Matrix3d dv_db_g_1 = dv_db_g + 0.5 * dt * (C * acc_S_x * cross + C_1 * acc_S_x * cross_1);
    dp_db_g += dt * dv_db_g + 0.25 * dt * dt * (C * acc_S_x * cross + C_1 * acc_S_x * cross_1);_