ethz-asl/okvis #44: Ending position result not repeatable

vincentcheng62, 2017-09-19:
For the same input (i.e. config file and dataset) to okvis_app_synchronous, I discover that the resulting position (i.e. the body frame position when progress=100%) have quite a large different for each single run.
Why is it so and how to fix it?
Thanks!

NikolausDemmel, 2017-09-19:
Without knowing okvis internals, what comes to mind are randomized algorithms (like ransac) and things that depend on timing (interaction between threads, borrow criteria based on execution time, which frames are processed and skipped)

vincentcheng62, 2017-09-25:
Thank you NikolausDemmel ! Your suggestion is very useful.
I fix the repeatibility problem by doing the following 3 steps:
(1) Set the time limit of ceres optimization to unlimit
(2) Set the bool RandomSeed=false of the 3 PosSacProblem headers
(3) Put the matching, optimizaing, frameconsumer and publisher threads into a single thread to fix their order of execution each time

vincentcheng62, 2017-09-26:
I also discover that increasing the ransac max iteration from 50 to 500 also improves repeatbility

NikolausDemmel, 2017-10-02:
Good to hear this has helped.

> However, some errors due to floating point accuracy still exists. But it is unlikely to be fixed at all

Just wanted to mention, that while floating point computation can lead to inaccuracies, by themselves they should still be deterministic, so that should not be a source of non-repeatability.

vincentcheng62, 2017-10-26:
Yes you are right NikolausDemmel that floating point inaccuracy still be deterministic.
Finally I discover that the multi-thread in DenseMatcher also cause repeatibility issue.
Setting NumOfThread=1 can solve the problem completely!

NikolausDemmel, 2017-10-26:
Great!

weblucas, 2018-04-13:
Without knowing the exact use that you need, these changes make the result completely different from the real-time solution in large datasets. An alternative is to perform comparisons running multiple times OKVIS on the same dataset.