ethz-asl/okvis #47: Does OKVIS support online calibration?

vincentcheng62, 2017-09-28:
As mentioned in the paper, online calibration produce fairly good result.
Does the OKVIS program support online calibration?

shenyingying, 2018-01-19:
I think no

weblucas, 2018-01-19:
yes. you need to change these parameters on the config yaml file

===============

camera_params:
    camera_rate: 20 # just to manage the expectations of when there should be frames arriving
    sigma_absolute_translation: 0.0 # The standard deviation of the camera extrinsics translation, e.g. 1.0e-10 for online-calib [m].
    sigma_absolute_orientation: 0.0 # The standard deviation of the camera extrinsics orientation, e.g. 1.0e-3 for online-calib [rad].
    sigma_c_relative_translation: 0.0 # The std. dev. of the cam. extr. transl. change between frames, e.g. 1.0e-6 for adaptive online calib (not less for numerics) [m].
    sigma_c_relative_orientation: 0.0 # The std. dev. of the cam. extr. orient. change between frames, e.g. 1.0e-6 for adaptive online calib (not less for numerics) [rad].
    timestamp_tolerance: 0.005 # [s] stereo frame out-of-sync tolerance