ethz-asl/okvis #57: Is there an implementation on iOS?

highlightz, 2017-12-19:


realitian, 2017-12-28:
@highlightz 
I am working on ios version.
btw, I had compared okvis with orb-slam2.
okvis used the akaze and  the akaze feature is robust than orb feature as I know.
but, I'm not sure about this okvis has no scale drift using imu...

highlightz, 2017-12-28:
@realitian This is good news there is someone working for this. Have you got any results on iphone device?   
On the other hand, I am a novice in ios development, and am in trouble moving all linux-based libraries to ios. Do you have any advice on how I should begin in order to make a live run on iphone device?

realitian, 2017-12-29:
@highlightz 
to build ios version, at first we should ensure that okvio works correctly for iphone recored camera feed and imu data as I thought.
so I am sharing iphone 7 plus config which i got calibrated.
let's make sure okvio work well for recored data from iphone.
cheers.
[iphone7p.yaml.zip](https://github.com/ethz-asl/okvis/files/1592809/iphone7p.yaml.zip)

closed 2018-04-16 by highlightz