ethz-asl/okvis #60: The app doesn't work well even on the given database

FlyAlCode, 2018-01-16:
When I run the app using the given database, it just given the following error:
*** Error in `/home/li/KDE/okvis-master/build/okvis_app_synchronous': double free or corruption (out): 0x00007fcb040101a0 ***

could you give me some advice?

weblucas, 2018-04-13:
The okvis_app_synchronous works fine. Which os and compiler are you using?

weblucas, 2018-04-16:
So, the software should work without any modification. We had over a dozen of students in my lab that did this installation on the same setup. My guess is that either you are running the software with wrong parameters or you have a dependence problem.  Try to use the okvis_ros. It has less chance of these problems.

Dragonbt, 2018-07-09:
i also have this issues
how can i do?