ethz-asl/okvis #63: build in tx2

caoyifeng001, 2018-03-06:
Hi:
I run BRISK/demo from https://www.doc.ic.ac.uk/~sleutene/software/brisk-2.0.4.zip in x86 and arm ,but i get different result in feature detection ,why? could help me.

EnTaroYan, 2018-04-14:
Hi caoyifeng001:
I build okvis in jetson tx2 and it has errors:
c++: error: unrecognized command line option ‘-mfpu=neon’
my jetpack version is 3.1 and gcc version is 5.4.0,Why?could help me
Thanks!

cosama, 2018-04-16:
caoyifeng001 already asked about the -mfpu=neon issue:

https://github.com/ethz-asl/okvis/issues/61

His thread was most likely closed because the solution can be found with a simple search and has been fixed in a stranded pull request:

https://github.com/ethz-asl/okvis/pull/21

Just adapt the CMakeList.txt according to that pull request and it should all work. Not sure if that will fix the BRISK issue too, but it will pull in a different version of it, so it might work differently.

caoyifeng001, 2018-04-17:
just delete it .

BirBikram, 2018-04-21:
Hi,
Not sure how to modify the CMakeList.txt. An example will be highly appreciated :grinning:

cosama, 2018-04-24:
This [CMakeLists.txt](https://github.com/ethz-asl/okvis/files/1944415/CMakeLists.txt) file should work on a TX2. It is necessary to use OpenCV2 for it to work though otherwise you run into this bug https://github.com/ethz-asl/okvis/issues/65.