ethz-asl/okvis #65: Frontend::initialiseBriskFeatureDetectors()  not compatible with OpenCV 3 on aarch64 architectures

cosama, 2018-04-16:
Compiling with the __ARM_NEON__ flag on currently doesn't work with OpenCV 3 and above. The only function affected seems to be 'Frontend::initialiseBriskFeatureDetectors()'. For example 'GridAdaptedFeatureDetector' has been removed and the way feature detectors are created has been changed.

The only solution for fixing this is to use OpenCV 2.

clement-joly, 2019-08-29:
@cosama I'm trying to make OKVIS run on ARM with OpenCV3.
I've managed to make the code compile but at runtime after processing half a dozen frames the program goes into a segmentation fault. See details [here](https://github.com/slightech/MYNT-EYE-OKVIS-Sample/issues/12).
Basically I've replaced the GridAdaptedFeatureDetector with a brisk::BriskFeatureDetector.
Any idea:
1. what could be triggering the segmentation fault?
2. if there might be another FeatureDetector that I could use that would be compatible with OpenCV3?

cosama, 2019-08-29:
@clement-joly I only filed this issue because I couldn't find an other way than using OpenCV2 to get this working. I actually haven't been using OKVIS for a long time, so I don't think I can help you much with this. Sorry.

clement-joly, 2019-08-30:
Thanks for the answer Marco. By any chance, do you happen to know of any other robust SLAM solution that integrates IMU?

cosama, 2019-08-30:
@clement-joly https://github.com/introlab/rtabmap is probably the best framework to do RGBD SLAM with your own system. If you like to have a ready to go solution, the [ZED-Mini](https://www.stereolabs.com/zed-mini/) and it's SDK does a pretty good job.

NikolausDemmel, 2019-08-30:
> By any chance, do you happen to know of any other robust SLAM solution that integrates IMU?

For RGB-D, you might wanna also have a look at https://github.com/ETH3D/badslam, which might be the state of the art in terms of accuracy. But I'm not sure if it works on ARM and what are the resource requirements.

VINSFusion seems to also have ROS integration and a stereo+IMU and mono+IMU mode: https://github.com/HKUST-Aerial-Robotics/VINS-Fusion

Also, there is [maplab / rovio(li)](https://github.com/ethz-asl/maplab)

For stereo+IMU, you might also wanna have a look at [basalt](https://gitlab.com/VladyslavUsenko/basalt/blob/master/doc/VioMapping.md#visual-inertial-odometry) (disclaimer: I'm one of the co-authors). Basalt's VIO is actually quite similar to OKVIS. The main difference is that it uses optical flow instead of feature matching. It is also a relatively efficient CPU implementation that on a multi-core PC can run much faster than realtime. However, I don't think it was tested on ARM so far. There is a demo application for live VO with the [Intel Realsense T256](https://gitlab.com/VladyslavUsenko/basalt/blob/master/src/rs_t265_vio.cpp), it is not a plug and play solution for your own camera. For example, it doesn't have ROS integration (yet).

clement-joly, 2019-09-25:
Thanks guys (sorry for the late answer), that's super helpful.