ethz-asl/okvis #67: How to get IMU-only trajectory?

thzt16, 2018-05-08:
I want to get the IMU-only trajectory, and use it to compare with ground-truth and Visual-Inertial output. How? Thank you !

weblucas, 2018-05-08:
OKVIS does not provide this, but for sure will be a terrible trajectory. I don't think you need to show that this is a bad idea. Just the noise and bias in the IMU are enough to give a completely wrong trajectory

thzt16, 2018-05-08:
Thanks for your answer! What I want to do is that, if the feature tracking lost, I want to optimize IMU data only, not include reprojection error term. This means that the IMU error term also has to be chsnged. Or just use the IMU integration result between two image frames to replace the joint optimization result. This idear right or wrong? Dose it work?

NikolausDemmel, 2018-05-08:
Just some anecdotal evidence: OKVIS seems to work fine if there is little-to-no visual information for a short period of time (at least if it has been well initialized). Check the results for sequences `slides{1,2,3}` of OKVIS on our dataset https://vision.in.tum.de/data/datasets/visual-inertial-dataset. Those include a section where the camera is sliding inside a closed tube w/o light. I didn't verify that all feature tracks are lost in those sections, but I strongly suspect it.