ethz-asl/okvis #75: stack smashing

WeskerRedfield, 2019-01-09:


WeskerRedfield, 2019-01-09:
When I ran okvis_app_asynchronous with an ASL dataset on ubuntu 16.04, I got the problem as following:

I0109 04:15:44.741714 20099 VioParametersReader.cpp:85] Opened configuration file: ./config/config_fpga_p2_euroc.yaml
I0109 04:15:44.742748 20099 VioParametersReader.cpp:173] imageDelay=0
I0109 04:15:44.761090 20099 VioParametersReader.cpp:275] Custom World frame provided T_Wc_W=
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
I0109 04:15:44.761153 20099 VioParametersReader.cpp:554] Found calibration in configuration file for camera 0
I0109 04:15:44.761176 20099 VioParametersReader.cpp:554] Found calibration in configuration file for camera 1
I0109 04:15:44.786283 20099 VioParametersReader.cpp:363] Radial tangential pinhole camera 0 with T_SC=
 0.0148655  -0.999881  0.0041403 -0.0216401
  0.999557  0.0149672  0.0257155  -0.064677
-0.0257744 0.00375619   0.999661 0.00981073
         0          0          0          1
I0109 04:15:45.017549 20099 VioParametersReader.cpp:363] Radial tangential pinhole camera 1 with T_SC=
 0.0125553  -0.999755  0.0182238 -0.0198436
  0.999599  0.0130119  0.0251588  0.0453689
-0.0253898  0.0179006   0.999517 0.00786212
         0          0          0          1
I0109 04:15:45.017580 20099 VioParametersReader.cpp:412] IMU with transformation T_BS=
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
I0109 04:15:46.077718 20099 okvis_app_synchronous.cpp:248] No. IMU measurements: 36820
I0109 04:15:46.089612 20099 okvis_app_synchronous.cpp:281] No. cam 0 images: 3682
I0109 04:15:46.102447 20099 okvis_app_synchronous.cpp:281] No. cam 1 images: 3682
W0109 04:15:46.138662 20113 Frontend.cpp:282] - Skipping propagation as only one IMU measurement has been given to frontend. Normal when starting up.
okvis_app_synchronous: malloc.c:3722: _int_malloc: Assertion `(unsigned long) (size) >= (unsigned long) (nb)' failed.
Aborted

I can't figure out how to fix it and will very appreciative of any suggestions.

WeskerRedfield, 2019-03-25:
Well, I've fixed this issue by setting USE_SYSTEM_CERES to ON in the CMakeLists.txt file, which means to use the ceres package that has already been installed on your system instead of the one okvis tries to download if you turn the option to OFF.
It appears there are some compatibility problems with SuiteSparse in terms of Ubuntu 16.04.
Anyway, I've found a solution to it and write it down here for others who may run into similar trouble.

quanzhpan, 2019-05-24:
SAME problem with you, how to deal with this please?

WeskerRedfield, 2019-05-28:
> SAME problem with you, how to deal with this please?

Check which version of SuiteSparse linked after your compilation and make sure it is a suitable one.