ethz-asl/okvis #82: make -j error

Astronirav, 2019-06-03:
Following error occurred when I hit the make -j8 command:
`/home/nsavaliya/git/okvis/build/brisk/src/brisk_external/src/brisk-feature-detector.cc: In function ‘void {anonymous}::RemoveInvalidKeyPoints(const cv::Mat&, std::vector<cv::KeyPoint>*)’:
/home/nsavaliya/git/okvis/build/brisk/src/brisk_external/src/brisk-feature-detector.cc:61:8: error: ‘function’ is not a member of ‘std’
   std::function<bool(const cv::KeyPoint& key_pt)> masking =
        ^~~~~~~~
/home/nsavaliya/git/okvis/build/brisk/src/brisk_external/src/brisk-feature-detector.cc:61:8: note: suggested alternative: ‘is_function’
   std::function<bool(const cv::KeyPoint& key_pt)> masking =
        ^~~~~~~~
        is_function
/home/nsavaliya/git/okvis/build/brisk/src/brisk_external/src/brisk-feature-detector.cc:61:17: error: expected primary-expression before ‘bool’
   std::function<bool(const cv::KeyPoint& key_pt)> masking =
                 ^~~~
/home/nsavaliya/git/okvis/build/brisk/src/brisk_external/src/brisk-feature-detector.cc:71:22: error: ‘masking’ was not declared in this scope
                      masking), keypoints->end());
                      ^~~~~~~
/home/nsavaliya/git/okvis/build/brisk/src/brisk_external/src/brisk-feature-detector.cc:71:22: note: suggested alternative: ‘mask’
                      masking), keypoints->end());
                      ^~~~~~~
                      mask
`

and also this:
`In file included from /home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/schur_eliminator_impl.h:64:0,
                 from /home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/generated/schur_eliminator_2_2_3.cc:48:
/home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/schur_eliminator.h: In member function ‘void ceres::internal::SchurEliminator<kRowBlockSize, kEBlockSize, kFBlockSize>::Init(int, const ceres::internal::CompressedRowBlockStructure*) [with int kRowBlockSize = 2; int kEBlockSize = 2; int kFBlockSize = 3]’:
/home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/schur_eliminator.h:267:10: error: ‘<anonymous>.ceres::internal::SchurEliminator<2, 2, 3>::Chunk::start’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   struct Chunk {
          ^~~~~
cc1plus: all warnings being treated as errors
internal/ceres/CMakeFiles/ceres.dir/build.make:2414: recipe for target 'internal/ceres/CMakeFiles/ceres.dir/generated/schur_eliminator_2_2_3.cc.o' failed
make[5]: *** [internal/ceres/CMakeFiles/ceres.dir/generated/schur_eliminator_2_2_3.cc.o] Error 1
make[5]: *** Waiting for unfinished jobs....
In file included from /home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/schur_eliminator_impl.h:64:0,
                 from /home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/generated/schur_eliminator_2_2_2.cc:48:
/home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/schur_eliminator.h: In member function ‘void ceres::internal::SchurEliminator<kRowBlockSize, kEBlockSize, kFBlockSize>::Init(int, const ceres::internal::CompressedRowBlockStructure*) [with int kRowBlockSize = 2; int kEBlockSize = 2; int kFBlockSize = 2]’:
/home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/schur_eliminator.h:267:10: error: ‘<anonymous>.ceres::internal::SchurEliminator<2, 2, 2>::Chunk::start’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   struct Chunk {
          ^~~~~
[ 56%] Building CXX object CMakeFiles/opengv.dir/src/math/arun.o
cc1plus: all warnings being treated as errors
internal/ceres/CMakeFiles/ceres.dir/build.make:2390: recipe for target 'internal/ceres/CMakeFiles/ceres.dir/generated/schur_eliminator_2_2_2.cc.o' failed
make[5]: *** [internal/ceres/CMakeFiles/ceres.dir/generated/schur_eliminator_2_2_2.cc.o] Error 1
In file included from /home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/schur_eliminator_impl.h:64:0,
                 from /home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/generated/schur_eliminator_2_2_4.cc:48:
/home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/schur_eliminator.h: In member function ‘void ceres::internal::SchurEliminator<kRowBlockSize, kEBlockSize, kFBlockSize>::Init(int, const ceres::internal::CompressedRowBlockStructure*) [with int kRowBlockSize = 2; int kEBlockSize = 2; int kFBlockSize = 4]’:
/home/nsavaliya/git/okvis/build/ceres/src/ceres_external/internal/ceres/schur_eliminator.h:267:10: error: ‘<anonymous>.ceres::internal::SchurEliminator<2, 2, 4>::Chunk::start’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   struct Chunk {
          ^~~~~
cc1plus: all warnings being treated as errors
internal/ceres/CMakeFiles/ceres.dir/build.make:2438: recipe for target 'internal/ceres/CMakeFiles/ceres.dir/generated/schur_eliminator_2_2_4.cc.o' failed
make[5]: *** [internal/ceres/CMakeFiles/ceres.dir/generated/schur_eliminator_2_2_4.cc.o] Error 1
CMakeFiles/Makefile2:117: recipe for target 'internal/ceres/CMakeFiles/ceres.dir/all' failed
make[4]: *** [internal/ceres/CMakeFiles/ceres.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/ceres_external.dir/build.make:111: recipe for target 'ceres/src/ceres_external-stamp/ceres_external-build' failed
make[2]: *** [ceres/src/ceres_external-stamp/ceres_external-build] Error 2
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ceres_external.dir/all' failed
make[1]: *** [CMakeFiles/ceres_external.dir/all] Error 2
`
Did anyone else face the same issue? How did you solve it?

Astronirav, 2019-06-04:
Issue solved. I'm using Ubuntu 18 and the default compile is g++7. Once I changed the compiler to g++6, it worked.

closed 2019-06-04 by Astronirav

jlussi, 2019-10-13:
Thank you for your response! I am having the same issue. I was wondering how your fix is  possible, are compilers not backwards compatible? Or was it the C-Standard?

chri321, 2024-05-31:
when i change gcc from version7 to version6 ，it didn't work .
and I try to fix schur_eliminator.h, as that struct
 Chunk {
  int start = 0; 
  int size = 0;
]
  problem fixed!!!!