ethz-asl/okvis #90: make -j8 error

charchemist, 2021-12-24:
I have noticed there has been the same error that appeared in orb-slam2 compiling, I wonder if there are some relationships between them. Can anybody solve this problem? thx

[ 60%] Building CXX object okvis_ceres/CMakeFiles/okvis_ceres.dir/src/Estimator.cpp.o
In file included from /usr/include/c++/9/map:61,
                 from /home/jcy/下载/okvis-master/build/include/ceres/ordered_groups.h:34,
                 from /home/jcy/下载/okvis-master/build/include/ceres/ceres.h:54,
                 from /home/jcy/下载/okvis-master/okvis_ceres/include/okvis/Estimator.hpp:48,
                 from /home/jcy/下载/okvis-master/okvis_ceres/src/Estimator.cpp:42:
/usr/include/c++/9/bits/stl_map.h: In instantiation of ‘class std::map<long unsigned int, okvis::MapPoint, std::less<long unsigned int>, Eigen::aligned_allocator<okvis::MapPoint> >’:
/home/jcy/下载/okvis-master/okvis_ceres/include/okvis/Estimator.hpp:563:19:   required from here
/usr/include/c++/9/bits/stl_map.h:122:71: **_error: static assertion failed: std::map must have the same value_type as its allocator
  122 |       static_assert(is_same<typename _Alloc::value_type, value_type>::value,
      |                                                                       ^~~~~**_
make[2]: *** [okvis_ceres/CMakeFiles/okvis_ceres.dir/build.make:232：okvis_ceres/CMakeFiles/okvis_ceres.dir/src/Estimator.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:559：okvis_ceres/CMakeFiles/okvis_ceres.dir/all] 错误 2
make: *** [Makefile:130：all] 错误 2

Avi241, 2022-02-05:
Please try it with g++-6.
https://linuxconfig.org/how-to-switch-between-multiple-gcc-and-g-compiler-versions-on-ubuntu-20-04-lts-focal-fossa for configuration of g++

EmilioOlivastri, 2022-11-30:
You can also solve this changing the typedef definition in the "FrameTypedefs.hpp" file in line 173 from :
- typedef std::map<uint64_t, MapPoint, std::less<uint64_t>, Eigen::aligned_allocator<MapPoint> > PointMap;
to 
- typedef std::map<uint64_t, MapPoint, std::less<uint64_t>, Eigen::aligned_allocator<std::pair<uint64_t const, MapPoint>> > PointMap;